intel/compile-time-init-build #64: Question: g++ 8 support

WojciechMigda, 2022-05-11:
Hi,

I have noticed that `g++` version 8.4.0 fails to compile specific example I have, which uses nexus with two different services, which require different invocation signatures. It compiles fine with `clang++` version 8, though.
Do you maybe know if this is a specific `g++` deficiency, or `cib` code could be modified in a way to handle this old g++ version as well?

Compiler error I am seeing is as follows:
```
/repo/github/cib-examples/ticker-extended/src/main.cpp:146:16:   required from here
/repo/github/cib-examples/external/cib/cib/nexus.hpp:50:53: error: invalid conversion from ‘void (*)(const spread_ind&)’ to ‘cib::traits::interface_t<publish_ticker>’ {aka ‘void (*)(std::__cxx11::basic_string<char>, const std::__cxx11::basic_string<char>&)’} [-fpermissive]
                     cib::service<clean_service_tag> = service_impl;
```

Code I am trying to build is here:
https://github.com/WojciechMigda/cib-examples/blob/main/ticker-extended/src/main.cpp

Thanks!

lukevalenty, 2022-05-11:
Thank you for submitting this issue! 😀

I don't typically use std::string in my code, so I didn't run into this issue. The problem is my code has it's own tuple that it uses instead of std::tuple. I wasn't fully qualifying cib::tuple and cib::make_tuple, this was causing some of the std equivalents to get called instead. I have a fix pending review.

WojciechMigda, 2022-05-11:
Thank you for such a prompt response! I am not sure this issue relates specifically to `std::string`. It more seems like `g++-8` tries to match feature invocation to a wrong service.

PR #66 (2022-05-11): fix cib::detail::name type string extraction algorithm
fixes #64
closed 2022-05-11 by lukevalenty