intel/compile-time-init-build #65: Support for services with r-value reference arguments

WojciechMigda, 2022-05-11:
Hi,
right now `cib` does not allow defining services which accept r-value arguments, what makes perfect sense since multiple features extending single service are allowed.
However, it would be of value to have an option (trait?) to modify service to allow only single listener, and with that, forwarding of arguments to a feature handler would be achievable.
Is that maybe something you have on your roadmap?

Thank you.

PS: I find `cib` very interesting and useful so far.

lukevalenty, 2022-05-11:
Hi @WojciechMigda, do you have a small example to illustrate what you want to do?

WojciechMigda, 2022-05-12:
Hi,

if I try to create service, which would require argument passed as r-value
```
struct publish_ticker : public cib::callback_meta<std::string &&>{};
```
with feature which matches above signature
```
struct ticker_processor
{
    constexpr static auto config =
        cib::config(
            cib::extend<publish_ticker>(
                [](std::string && symbol)
                {
                    fmt::print("ticker_processor: {}\n", symbol);
                }
            )
        );
};

```
then, if I attempt to call the service passing r-value argument
```
cib::service<publish_ticker>("symbol");
```
Then the compiler (clang 8) complains with the error below:
```
cib/callback.hpp:62:22: error: rvalue reference to type 'basic_string<...>' cannot bind to lvalue of type 'basic_string<...>'
                func(args...);
                     ^~~~
```
If I understand the gist of `cib`'s inner workings, then it keeps track of how many features were specified for a given service. With that, when a service call point is encountered, `cib` will be able to determine at compile time that it could forward the arguments.

I am not that good meta programmer (if at all), but the `run` function in `callback.hpp` could have a check for `num_funcs`, allowing forwarding when it equals `1`.
```
            if constexpr (num_funcs == 1) {
                CIB_CONSTEXPR auto func = handler_builder.funcs[0];
                func(std::forward<ArgTypes>(args)...);
            }
            else
            {
                detail::for_each(num_funcs, [&](auto i){
                    CIB_CONSTEXPR auto func = handler_builder.funcs[i];
                    func(args...);
                });
            }
```
Do you think it makes sense?

lukevalenty, 2022-05-12:
This was a great explanation, thank you!

It seems like there should be a way to make rvalue references work for multiple callbacks, but it might require a copy to made. For the case where only one callback is registered, no copies would be made and the arguments could be forwarded.

lukevalenty, 2022-05-12:
By the way, in your example, it might be a good idea to use std::string_view instead. String views don't allocate any dynamic memory and are well suited to embedded or high-performance applications.

lukevalenty, 2022-05-13:
I'm leaning towards not supporting rvalue references as function arguments for callback services. An important attribute of services and features is that they are independent. Once we use rvalue references and forward/move objects, the different services and features become more coupled.

WojciechMigda, 2022-05-14:
Is your reservation limited to this particular proposed solution hint I included or supporting rvalue references as a whole?

lukevalenty, 2022-05-14:
It's for rvalue references in general. If we are passing objects from one entity to another, and we don't know how many receivers there might be, then it doesn't make sense to pass ownership of the object.

If we limit the receivers to one, then we are breaking the contract of keeping components decoupled from each other. The moment another component wants to register for that service, compilation will fail.

closed 2022-05-31 by lukevalenty