intel/compile-time-init-build #71: Stateful nexus with dependencies

WojciechMigda, 2022-05-23:
Hi,

it occured to me yesterday that it would be a nice feature to have stateful nexus, which, in spirit of DI, would then be able to inject extra arguments to handlers registered through features.

What I have in mind is something that Kris Jusiak's library `SML` already has. https://github.com/boost-ext/sml
Specifically, in Kris'  library state machine instance can be instantiated with extra dependencies. Then, through 'smart' resolution it is able to pass these dependencies to actions which are part of state machine's transitions.
Such basic example is presented in `SML`'s documentation, specifically, [Section 5: _Create a state machine_](https://boost-ext.github.io/sml/tutorial.html).

In case of `CIB`, it would allow to pass user's dependencies to created nexus:

`cib::nexus<nexus_project> nexus{Type1: dependency1, Type2: dependency2, ...};`

Then, `CIB` when it matches handlers registered against service signature, it would also allow handler's signatures which arguments are superset of those specified in service's definition, extended with types which can be matched against dependencies passed to instantiated nexus. For example:

    struct service : public cib::callback_meta<Payload const &>{};`
    
    struct feature
    {
        static auto constexpr config = cib::config(
            cib::extend<service>(
                [](double x, Payload const & p)
                {
                    // do sth with injected x and passed p
                }
            )
        );
    };

    cib::nexus<nexus_project> nexus{87.0};

In the above `service` is declared to accept const reference to `Payload`, but then we'd attempt to declare feature which is attached to a handler which accepts `Payload` _and_ a `double`. Since nexus is instantiated with a `double` as one of dependencies passed by a user it would allow such feature and pass both arguments - one originating from the point of invocation of the service through the nexus, and the other from the pool of dependencies which are part of nexus' state.

I believe this would be huge undertaking to introduce such enhancement, but given that Kris' `SML` library already has it, maybe it could be used as a guidance?

Thanks,
Wojciech

lukevalenty, 2022-05-25:
This is a fantastic feature suggestion! I've looked at @krzysztof-jusiak's DI library and found that it didn't work in a constexpr context for myself. Maybe there are some changes that could be made.

I would want to be able to inject subclasses of a base interface (virtual function overriding.)

I would also want to inject arbitrary types into template functions.

Thanks for this suggestion @WojciechMigda.

krzysztof-jusiak, 2022-05-26:
I think that would be nice. I'm happy to experiment and see what can be done.

antonysigma, 2025-02-10:
Just curious... are we referring to runtime dependency injection at pre-init stage or post-init stage?

The title of this ticket seemed to indicate pre-init DI, but I have seen a lot real-world use-case that requires DI after runtime-init. 

For example,  runtime DI driven by external (e.g. UART) commands:

```c++
struct ExternalPumpBoardConnection {
   uint8_t i2c_addr{};
};

constexpr size_t MAX_PUMPS = 256;
using PumpBoardList = std::stack_space_vector<ExternalPumpBoardConnection, MAX_PUMPS>;

int main() {
    cib::nexus<bringup_project> nexus{};
    nexus.service<RuntimeInit>();

   // Post-init device bring up
   PumpBoardList pumps;
   bool finished = false;
   do {
        finished = console<UartDevice>::onPumpConnectionCommand(
        [](const PumpConnection pump) {
            pumps.emplace_back(pump);
       });
   } while(!finished);

  cib::nexus<runtime_project> new_nexus{pumps};

  new_nexus.service<RuntimeInit>();
  for(;;) {
      new_nexus.service<MainLoop>();
  }
}
```