intel/compile-time-init-build #221: Logger uses dynamic memory (std string)

Thurnheer, 2022-11-23:
fmt::format returns a string, so every Log, Trace etc will allocate memory

Thurnheer, 2022-11-29:
I think commit 771680f014421c219038d77335c7aa8639adadb4 proves that the CIB_TRACE and log macros will allocate dynamic memory through new.
This is a bit problematic for us as that the message handler uses those macros and therefor allocates memory on every message handling invocation.

lukevalenty, 2022-11-29:
Correct! Thanks for bringing this up along with your pull request to fix it. 😄 

I had intended for the fmt logger to only be used for testing, not on embedded systems due to code size and performance. But there are perfectly valid use cases for it in all but the smallest of embedded systems. 

The mipi logger is the one I was focusing on for embedded, but it requires a bit more infrastructure to support. Now we have two options that both work for embedded. I'm closing this out as fixed with the merge of #222

closed 2022-11-29 by lukevalenty