intel/compile-time-init-build #261: add support for field aliases in msg library

lukevalenty, 2023-02-13:
There are cases where we have a single logical field name that has multiple different locations depending on what message type it is in. This issue is for an enhancement to alias a single field name to multiple field types depending on which message it is in.

herculanodavi, 2023-08-08:
Would it be ok to have a type that stores only the width and value type of the field in, say, `field_meta` and to have a logic to compose the type with location information? Maybe this could be done:

```c++
using b_x = field_meta<decltype('x'_sc), 16, int16_t>;
using b_y = field_meta<decltype('y'_sc), 16, int16_t>;
using b_z = field_meta<decltype('z'_sc), 16, int16_t>;

using data_msg = message_base<
  decltype('data'_sc),
  3,
  b_x::at<0, 15, 0>, // type alias to a `msg::field`.
  b_y::at<0, 31, 16>,
  b_z::at<1, 15, 0>
>;
``` 

Happy to make a PR for that if approved.

lukevalenty, 2023-08-08:
I like this idea! I have a couple thoughts:
1. The final API for this should be designed to also work with disjoint fields.
2. The `at<...>` should use the `field_meta` width by default.

I would also like to see what @bdeane-intel and @mjcaisse-intel have to think about this.

bdeane-intel, 2023-08-08:
My thoughts:

I like separating field info from where the field is located. (Can we find a better name than `field_meta`?)

I'd like disjoint_field and field to be the same thing, really (although field could be specialized differently for the single-span case).

Let's write a concept for the field interface?

I like the idea of a locator type that can be expressed a few different ways, e.g.
- dword offset + lsb offset (e.g. <1, 0>)
- lsb offset only (e.g. <32> would be equivalent to <1, 0>)

And yes, the locator should not include the size information of the field (directly or indirectly), given that that's already in the field info.

We might also default adjacent message fields to being contiguous and the default start would be with bit 0, so a message definition in that case could simply list fields without locator information.

herculanodavi, 2023-08-08:
> Can we find a better name than `field_meta`?

I thought of `field_base` or `field_info` but I have no strong feelings for any.

> Let's write a concept for the field interface?

Are there guidelines on how to name concepts? I see the current concepts in this repo are adverbs, but is this a rule? I ask this to differentiate between the field concept and type.

> We might also default adjacent message fields to being contiguous and the default start would be with bit 0, so a message definition in that case could simply list fields without locator information.

So something like:
```c++
using b_x = field_meta<decltype('x'_sc), 16, int16_t>;
using b_y = field_meta<decltype('y'_sc), 16, int16_t>;
using b_z = field_meta<decltype('z'_sc), 16, int16_t>;

using data_msg = message_base<
  decltype('data'_sc),
  3,
  b_x,         // this assumes lsb=0
  b_y,         // this assumes lsb=16
  b_z::at<35>, // For non-trivial lsb offset
>;
```

------
If the location is an alias inside `field_meta`, could we be making extendability difficult by hurting SRP? If so, wouldn't it be better to have these as free aliases, like `offset<b_z, 35>` instead of `b_z::at<35>`?

bdeane-intel, 2023-08-08:
I quite like `field_spec`.

Then maybe a `field_desc` is a `field_spec` with a `field_locator`. And several `field_spec`s are the arguments to `message_base`.

`field_locators` can come in absolute or relative flavours, and the default could be a relative one that has a zero offset.

Still kind of blue-skying here but I think this could work quite well.

lukevalenty, 2023-08-08:
To add, if we make these kinds of changes, we can still present a compatible API by implementing the existing field types using the new classes.

elbeno, 2023-11-10:
I think we can close this issue now - @lukevalenty ?

closed 2023-11-10 by lukevalenty