intel/compile-time-init-build #315: Accept non-narrowing conversions in the `msg::message_base` constructor

herculanodavi, 2023-06-28:
In the case of communicating with devices with 8-bit long registers using SPI or I2C, most HAL outputs are arrays of `std::uint8_t`, which could be copied safely to the `msg::message_data` type, `std::uint32_t`.

However, the constructor of `msg::message_base` asserts the `R::value_type` to be `std::uint32_t`, so this cast is not possible.  
```c++
template <detail::range R> explicit constexpr message_base(R const &r) {
    static_assert(std::is_same_v<typename R::value_type, std::uint32_t>);
    resize_and_overwrite(*this, [&](std::uint32_t *dest,
                                    std::size_t max_size) {
        std::copy_n(std::begin(r), std::min(std::size(r), max_size), dest);
        return NumDWordsT;
    });
}
```

Would it make sense to either:
1. Change the `detail::range` concept to add the requirement `std::convertible_to<std::uint32_t>` and remove the `static_assert`?
2. Change the `std::is_same_v` inside the `static_assert` to `std::is_convertible<typename R::value_type, std::uint32_t>`?

I'd be happy to make a PR for that.

herculanodavi, 2023-06-29:
Thank you! This solves it for now, though using other types as storage would also be great. 

I see the issue is still open, so should I close it or would it be closed when the todo is done?

bdeane-intel, 2023-06-29:
Let's leave it open as a TODO. I think @mjcaisse-intel already has an implementation, so hopefully we can get that into cib soon.

PR #413 (2023-11-07): :art: :recycle: Refactor messages
Messages are reworked:
 - separate definition from storage
 - owning, mutable view and const view types
 - construction from arrays and/or by setting fields
 - get and set fields by name

Handlers, matchers and callbacks don't incur any copying any more; message views
can be constructed cheaply over raw arrays.

Closes: #315
Closes: #394
Closes: #395
Closes: #398
closed 2023-11-07 by lukevalenty