intel/compile-time-init-build #330: Trouble understanding more complex use case

jferch, 2023-08-10:
Hi everyone,

thanks for the interesting library, I really appreciate the design concept and philosophy!

I'm struggling a bit however how one would build up a more complicated example which require additional dependencies (maybe closely related to this [issue](https://github.com/intel/compile-time-init-build/issues/71 )?). The examples I've seen are either using pure functions or utilize std::cout/global variables. I unfortunately can't seem to grasp the concept from the documentation, tests or the presentations but quite sure I'm missing something.

I don't believe passing them into the lambdas with captures would be possible due to func_ptr_t being a regular function pointer? Would I need to add depending types as a static members but then this would make testing/mocking individual parts quite difficult?

Thanks so much in advance!
Johannes

lukevalenty, 2023-08-14:
I think some more practical examples would be very helpful for you. We have a large code-base using previous iterations of this library, and a new one using this version now. Unfortunately we can't share any of the code that uses this library. I will think about some more practical and relevant examples we can create to help explain how this works. 🤔

jferch, 2023-08-18:
@lukevalenty Yes, I figured. That would be great, thanks! :) Have a good weekend