intel/compile-time-init-build #457: 'msg::handler' interface: allow messages to pass by value

lukevalenty, 2023-12-23:
Some embedded architectures, ABIs, messages sizes, and compilers may benefit from passing messages by value rather than reference.

Passing by value in some cases allows the entire message to stay in registers. This saves storing the message to memory and retrieving it from memory for each callback.

elbeno, 2024-05-21:
Clang has the builtin function `__can_pass_in_regs(type)`  which might be useful, but I don't know if there is a similar intrinsic for GCC or whether it is widely applicable.

elbeno, 2024-10-30:
@lukevalenty Is this addressed by https://github.com/intel/compile-time-init-build/pull/615 ?

lukevalenty, 2024-11-06:
Yes, I think it is addressed by #615.