intel/compile-time-init-build #506: Can't cross compile for SAM4L-EK with arm-none-eabi-g++

rpreddyhv, 2024-02-26:
I could compile the cib repo with g++ v13.1.0 on Ubuntu 22.04 in VMWare virtual machine. However, when I tried to cross-compile the same for SAM4l-EK (SAM4LC4C - Arm Cortex M4) with arm-none-eabi-g++ v10.3.1, it was failing to find definitions for SA_ONSTACK.

```
[ 30%] Building CXX object _deps/catch2-build/src/CMakeFiles/Catch2.dir/catch2/internal/catch_fatal_condition_handler.cpp.obj
<project_path>/build/_deps/catch2-src/src/catch2/internal/catch_fatal_condition_handler.cpp: In member function 'void Catch::FatalConditionHandler::engage_platform()':
<project_path>/build/_deps/catch2-src/src/catch2/internal/catch_fatal_condition_handler.cpp:226:23: error: 'SA_ONSTACK' was not declared in this scope; did you mean 'SS_ONSTACK'?
  226 |         sa.sa_flags = SA_ONSTACK;
      |                       ^~~~~~~~~~
      |                       SS_ONSTACK
make[2]: *** [_deps/catch2-build/src/CMakeFiles/Catch2.dir/build.make:916: _deps/catch2-build/src/CMakeFiles/Catch2.dir/catch2/internal/catch_fatal_condition_handler.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:5788: _deps/catch2-build/src/CMakeFiles/Catch2.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

**My tool chain configuration:** (_local_toolchains/arm-none-eabi.cmake_)
```
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)

set(CMAKE_C_COMPILER "arm-none-eabi-gcc")
set(CMAKE_CXX_COMPILER "arm-none-eabi-g++")
set(CMAKE_CXX_FLAGS "-mthumb -ffunction-sections -fdata-sections -Wall")

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 20)

set(CMAKE_EXE_LINKER_FLAGS "--specs=nosys.specs" CACHE INTERNAL "")

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
```

**Commands used:**
```
mkdir build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE:PATH=../local_toolchains/arm-none-eabi.cmake ..
make
```

**Compiler version:**
```
arm-none-eabi-g++ --version
arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

elbeno, 2024-02-26:
- the `main` branch supported GCC versions are 12 and later at the moment
- the error is in Catch2, not CIB. Tests won't cross-compile for baremetal in general (and they don't need to) because test frameworks are going to rely on OS stuff. Do the examples compile?

rpreddyhv, 2024-02-26:
I could cross-compile hello_world with some udpates to toolchains.
```
compile-time-init-build/examples/hello_world$ file hello_world
hello_world: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped
```

So, I don't need to cross-compile complete repo for target.

I will include the repo as module and build on it.

Thank you @elbeno.

rpreddyhv, 2024-02-26:
Reopening this issue as I couldn't cross-compile latest cib with updated toolchain. Earlier I tried to compile hello_word example which was using old cib and toolchain was also gcc version 10.3.
```
[ 25%] Building CXX object CMakeFiles/sam4l-ek.dir/main.cpp.obj
In file included from <project_path>/examples/sam4l-ek/build/_deps/cib-src/include/interrupt/dynamic_controller.hpp:3,
                 from <project_path>/examples/sam4l-ek/build/_deps/cib-src/include/interrupt/manager.hpp:4,
                 from <project_path>/examples/sam4l-ek/build/_deps/cib-src/include/cib/cib.hpp:44,
                 from <project_path>/examples/sam4l-ek/main.cpp:3:
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:22:33: error: 'mutex' in namespace 'std' does not name a type
   22 | template <typename Mutex = std::mutex> class standard_policy {
      |                                 ^~~~~
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:16:1: note: 'std::mutex' is defined in header '<mutex>'; did you forget to '#include <mutex>'?
   15 | #include <conc/concepts.hpp>
  +++ |+#include <mutex>
   16 | 
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:53:78: error: class template argument deduction failed:
   53 | template <typename...> inline auto injected_policy = detail::standard_policy{};
      |                                                                              ^
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:53:78: error: no matching function for call to 'standard_policy()'
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:22:46: note: candidate: 'template<class Mutex> standard_policy()-> conc::detail::standard_policy<Mutex>'
   22 | template <typename Mutex = std::mutex> class standard_policy {
      |                                              ^~~~~~~~~~~~~~~
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:22:46: note:   template argument deduction/substitution failed:
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:22:46: note: candidate: 'template<class Mutex> standard_policy(conc::detail::standard_policy<Mutex>)-> conc::detail::standard_policy<Mutex>'
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:22:46: note:   template argument deduction/substitution failed:
<project_path>/examples/sam4l-ek/build/_deps/cpp-baremetal-concurrency-src/include/conc/concurrency.hpp:53:78: note:   candidate expects 1 argument, 0 provided
   53 | template <typename...> inline auto injected_policy = detail::standard_policy{};
      |                                                                              ^
make[2]: *** [CMakeFiles/sam4l-ek.dir/build.make:76: CMakeFiles/sam4l-ek.dir/main.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:187: CMakeFiles/sam4l-ek.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```
**updated toolchain version**:
```
$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
**sam4l-ek/main.cpp**:
```
#include <cib/cib.hpp>
int main() {
    return 0;
}
```
**sam4l-ek/CMakeList.txt**:
```
cmake_minimum_required(VERSION 3.16)
project(sam4l-ek LANGUAGES CXX)
add_executable(sam4l-ek main.cpp)

if(CMAKE_SOURCE_DIR STREQUAL CMAKE_CURRENT_SOURCE_DIR)
    set(CIB_VERSION "c3ebc7d")
    if(CPM_DOWNLOAD_VERSION)
        cpmaddpackage("gh:intel/compile-time-init-build#${CIB_VERSION}")
    else()
        include(FetchContent)
        FetchContent_Declare(
            cib
            GIT_REPOSITORY https://github.com/intel/compile-time-init-build.git
            GIT_TAG ${CIB_VERSION})
        FetchContent_MakeAvailable(cib)
    endif()
endif()

target_link_libraries(sam4l-ek cib)
```
**local_toolchains/arm-none-eabi.cmake**:
```
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)

set(TOOLCHAIN_ROOT "/usr")
if(DEFINED ENV{TOOLCHAIN_ROOT})
    set(TOOLCHAIN_ROOT $ENV{TOOLCHAIN_ROOT})
endif()

set(CMAKE_CXX_COMPILER "${TOOLCHAIN_ROOT}/bin/arm-none-eabi-g++")
set(CMAKE_CXX_FLAGS "-mthumb -ffunction-sections -fdata-sections -Wall -std=c++20")
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_EXE_LINKER_FLAGS "--specs=nosys.specs -mcpu=cortex-m4 -march=armv7-m" CACHE INTERNAL "")

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
```

**commands**:
```
setenv TOOLCHAIN_ROOT=<toolchain_path>
cd examples/sam4l-ek
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE:PATH=../../local_toolchains/arm-none-eabi.cmake ..
make
```
```
I think when I compile using native compiler it could find mutex definition in std namespace. However, when I try to cross-compile for arm it couldn't find the mutex definition. How do I resolve this?

Thurnheer, 2024-02-26:
Had also this problem... seems to be an issue with a definie in cpp-barmetal-concurrency.
I could solve it by using -DSIMULATE_FREESTANDING in the cmake

elbeno, 2024-02-27:
Yes I think this is a general issue with toolchains that provide headers (for some reason) but those headers don't contain what's in the standard. If `__has_include(<mutex>)` is true but `std::mutex` doesn't exist, I'm not sure what else to do... for things that have feature test macros I generally use those, but even there toolchains lie sometimes...

elbeno, 2024-02-27:
Freestanding support in C++ might be OK around C++29 timeframe. Until then I'm afraid things like this are a bit ad-hoc. Open to suggestions.

rpreddyhv, 2024-03-21:
I could build cib for a basic LED & Push Button example for SAM4L-EK Board. Tested on the board.

Posted it to github https://github.com/rpreddyhv/cib-embedded

closed 2024-03-21 by rpreddyhv