intel/compile-time-init-build #653: Aggregate multiple messages into one

oxidase, 2024-11-27:
I would like to define a set of messages with a header and CRC field but multiple payload definitions. 
Would it be possible in the current code base?

Something like with a `msg::combine` function similar to `msg::extend`:
```
using header_def = msg::message<"header", payload_type_f, ... >;
using payload1_def = msg::message<"payload1",  ...>;
using payload2_def = msg::message<"payload2",  ...>;
using crc_f = msg::field<"crc", std::uint16_t>::located<at{14_msb, 0_lsb}}>;

using packet1_def = msg::combine<  
   header_def,
   payload_type_f::with_required<1>,
   payload1_def, 
   crc_f
>;

using packet2_def = msg::combine<  
   header_def,
   payload_type_f::with_required<2>,
   payload2_def, 
   crc_f
>;
```

The packet layout should be
```
| header | payload bit addresses + bit size of header | crc 15 bit + bit size of header and payload|
```

elbeno, 2024-11-27:
I think your `packet1_def` is equivalent to:
```cpp
using packet1_def = msg::extend<
    header_def, 
    "packet1", 
    payload_type_f::with_required<1>,
    crc_f,
    [payload1_def_fields...]>;
```
So it should be easy to write an alias template to achieve that?

oxidase, 2024-11-28:
@elbeno yes, that what I would like to achieve. Let me be specific with a failing test case

```
TEST_CASE("combined messages", "[message]") {
    using ext_id = field<"ext.id", std::uint32_t>::located<at{0_dw, 31_msb, 24_lsb}>;
    using ext_field1 = field<"ext.f1", std::uint32_t>::located<at{0_dw, 15_msb, 0_lsb}>;
    using payload_defn = message<"payload", ext_id, ext_field1>;

    using packet_defn = msg::extend<
        msg_defn
        , "packet"
        , id_field::with_required<0xaa>
        , payload_defn::fields_t
        >;

    CHECK(packet_defn::size<std::uint8_t>::value == 11);
}
```

1. What would a simplest way to unpack `stdx::type_list<Fields...>` fields from `payload_defn::fields_t`?
2. `ext_id` has 0 dwi in the  payload definition and it shall have 7 bi in the packet definition.

elbeno, 2024-11-28:
1. Can be done with eg. `stdx::apply_sequence` or similar metaprogramming pattern.
2. Has no library machinery: generally you are talking about automatically locating fields in messages and that isn't supported today. However you could take a list of fields, either without locations and pack them, or with locations that  you offset by the header size, if that is the use case.

oxidase, 2024-12-01:
1. Good point!
2. I got to the implementation for my use-case. For a message test
```
TEST_CASE("combined messages", "[message]") {
    using src = field<"src", std::uint32_t>::located<at{0_dw, 31_msb, 24_lsb}>;
    using dst = field<"dst", std::uint32_t>::located<at{0_dw, 15_msb, 0_lsb}>;
    using payload_defn = message<"payload", src, dst>;

    using packet_defn = combine_messages_t<
        "packet"
        , msg_defn
        , payload_defn
        , id_field::with_required<0xaa>
        >;

    CHECK(msg_defn::size<std::uint32_t>::value == 2);
    CHECK(payload_defn::size<std::uint32_t>::value == 1);
    CHECK(packet_defn::size<std::uint32_t>::value == 3);

    boost::mp11::mp_for_each<packet_defn::fields_t>([](auto F) {
        std::cout << typeid(decltype(F){}).name() << "\n";
    });

    using packet_const_view = msg::const_view<packet_defn>;

    auto const arr = typename packet_defn::default_storage_t{0x8000'ba11, 0x0042'd00d, 0x12345678};
    auto const view = packet_const_view{arr};
    fmt::print("id  0x{:02x}\n", view.get("id"_f));
    fmt::print("src 0x{:02x}\n", view.get("payload.src"_f));
    fmt::print("dst 0x{:04x}\n", view.get("payload.dst"_f));
}
```
the test prints
```
msg::detail::field_t<"f1"_sc, unsigned int, msg::detail::with_default<unsigned int, 0u>, match::always_t, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)0, (msg::msb_t)15}>
msg::detail::field_t<"id"_sc, unsigned int, msg::detail::with_const_default<unsigned int, 170u>, msg::rel_matcher_t<std::__1::equal_to<void>, msg::detail::field_t<"id"_sc, unsigned int, msg::detail::with_default<unsigned int, 0u>, match::always_t, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)0, (msg::msb_t)31, (msg::lsb_t)24}>, 170u>, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)0, (msg::msb_t)31, (msg::lsb_t)24}>
msg::detail::field_t<"f3"_sc, unsigned int, msg::detail::with_default<unsigned int, 0u>, match::always_t, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)1, (msg::msb_t)15, (msg::lsb_t)8}, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)1, (msg::msb_t)7}>
msg::detail::field_t<"f2"_sc, unsigned int, msg::detail::with_default<unsigned int, 0u>, match::always_t, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)1, (msg::msb_t)23, (msg::lsb_t)16}>
msg::detail::field_t<"payload.dst"_sc, unsigned int, msg::detail::with_default<unsigned int, 0u>, match::always_t, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)2, (msg::msb_t)15}>
msg::detail::field_t<"payload.src"_sc, unsigned int, msg::detail::with_default<unsigned int, 0u>, match::always_t, msg::at<msg::dword_index_t, msg::msb_t, msg::lsb_t>{(msg::dword_index_t)2, (msg::msb_t)31, (msg::lsb_t)24}>
id  0x80
src 0x12
dst 0x5678
```

If it is interesting I can create a PR for this feature

lukevalenty, 2024-12-10:
@elbeno's #655 PR implements this.

closed 2024-12-10 by lukevalenty