intel/compile-time-init-build #713: Support for 64-bit and array fields in CIB

shaulnv, 2025-03-21:
My question is on the CIB message component.
- Does the library support (up to) 64-bit fields? 
- Does it support fields of arrays of u8/u16/u24/... or arrays of other defined fields?  

From what I found in the documentation, CIB only supports bit fields of <= 32-bits, contained in 32-bit aligned boundaries:  
[image]

Is that the case?

Thanks for this cool library, and the great talks you gave about it :-) 
Shaul

lukevalenty, 2025-03-21:
Hi @shaulnv! The underlying storage of messages is `std::uint32_t`, but we can still support fields from 1-bit up to 64-bits in size. The message library handles packing and unpacking the fields.

However, it does not currently support arrays of fields. Are you looking for support for an array of fixed length, or a dynamic length based on some size field in the message?

shaulnv, 2025-03-21:
It is actually both :-)
I know the capacity statically, but the size is dynamically.
I think that from CIB POV, this is just a static array field, where the dynamic part is done in runtime, as it is not reflected in the message layout, right?

```cpp
struct child_qp {
  u_int32_t child_qpn;
  u_int8_t child_index;
};

struct message {
  uint8_t num_of_children;
  child_qp child_qp[44];
};
```

mjcaisse-intel, 2025-06-13:
Hi @shaulnv -

That documentation comment should be clarified. The underlying storage defaults to `uint32_t`; however, you can use the `custom_storage_t` since this commit: https://github.com/intel/compile-time-init-build/commit/8e676e08f1ad87386eae3f4125cffff0ef052dfe  (see included test)

You can also construct message views over existing memory of varying types: https://github.com/intel/compile-time-init-build/blob/main/test/msg/message.cpp#L132-L137

64-bit fields are supported: https://github.com/intel/compile-time-init-build/blob/main/test/msg/field_extract.cpp#L96-L103

As you have mentioned, message layout is static in the library. Looks like we need to improve the docs in this area. Thank you for the feedback!
michael