intel/compile-time-init-build #723: Roadmap to support AVR/8-bit chipset

antonysigma, 2025-04-17:
(Summarizing the discussions at https://github.com/intel/compile-time-init-build/discussions/105 ... Please feel free to revise the roadmap below.)

1. [ ] Wait for ISO C++26 standard to publish, which requires the standard library `<stdexcept>` to be part of the freestanding library;
2. [ ] Work with team behind the project [avr-libstdcpp](https://github.com/modm-io/avr-libstdcpp) to include a subset of `<stdexcept>` features. https://github.com/modm-io/avr-libstdcpp?tab=readme-ov-file#notable-adaptions-and-limitations
3. [ ] If step 2 is blocked, convince the fmtlib team to remove the line `#include <stdexcept>` in the header-only library, or make it an optional `#include` similar to how `#include <fmt/compile.h>` is made optional.
4. [ ] If step 2 and 3 are blocked, convince the `compile-time-init` team to make `#include <fmt/format.h>` optional.
5. [ ] If step 2, 3, and 4 are all "no go", the feature is clearly stuck in a multi-team, multi-project stalemate. maybe we should give up on Microchip/Atmel AVR chipset support in the `compile-time-init` framework. (After all, the limited 16-bit address space, and the Harvard architecture may not justify `cib`'s complexity.) 
6. [X] Or, use an outdated version without the latest features, e.g. RLE encoding of the command router, i.e. right before #338 .

elbeno, 2025-04-17:
👍 

IMO we are in the middle of a re-reckoning for the avr-libstdcpp claim "Some parts of the C++ standard library are not well suited for tiny bare-metal systems." It may still be true, but to a much lesser extent given the amount of stuff that is becoming usable purely at compile-time.

We don't really have to wait for C++26 for avr-libstdcpp to implement `<string>` (given `std::string` is usable constexpr in C++20) and once you have `<string>`, `<stdexcept>` needn't be far behind.

antonysigma, 2025-12-05:
Thanks @elbeno for the guidance. I probably will need to take a deep dive into the `.query()` design pattern, as well as `get_string_id`. I don't know enough to compile the example code `examples/log/binary_custom/logger.hpp` with `avr-gcc`.

So far, the concept `valid_query_for` always evaluate to false, which in turn triggers the following error:

```c++
../include/log/catalog/encoder.hpp:77:19: error: no class template named 'fn' in 'struct logging::binary::detail::to_message_t<decltype (fr.str), -1>'
   77 |             using Message = typename decltype(builder)::template convert_args<
      |                   ^~~~~~~
```

If you wish, I can file a separate ticket for the logging + avr-gcc integration issue. Complete log:
https://github.com/antonysigma/compile-time-init-build/actions/runs/19955616944/job/57224047455#step:12:68